harbaum/I2C-Tiny-USB #8: Linux GPIOs with I2C-GPIO expanders (PCF8574)

zoobab, 2018-09-05:
Hi,

I have flashed a Digispark with the i2c-tiny-usb firmware, connected a PCF8574, and under linux, it seems you can obtain those 8 GPIOs exported in /sys/class/gpio.

However, I cannot manage to see them there.

Do I need those 10K resistors somewhere? I have the impression the PCF does not receive 5V on its VCC pin, so I suspect this might be the reason. Do you have a better picture or a schema where to place those 10K resistors?

That would be great to document the usage of those, as GPIOs over USB are pretty hard to get (out-of-tree driver, etc...):

http://www.zoobab.com/ch341-usb-spi-i2c-uart-isp-dongle#toc15

zoobab, 2018-09-06:
Should the Digispark adapter appear in the chain when I do an i2cdetect?

drawkula, 2018-09-06:
I just flashed a Digispark...
```
$ lsusb | fgrep i2c
Bus 002 Device 010: ID 0403:c631 Future Technology Devices International, Ltd i2c-tiny-usb interface
$ i2cdetect -l
```
No output. Probably the `i2c-dev` module is missing...
```
$ sudo modprobe i2c-dev 
$ i2cdetect -l | fgrep tiny
i2c-10  unknown         i2c-tiny-usb at bus 002 device 010      N/A
```
That's how it looks on my Debian9 notebook. I cannot test more because finding some i2c chips in my electronic sediments would take longer than I got time atm...

zoobab, 2018-09-06:
Will try loading that module this evening...

harbaum, 2018-09-06:
Indeed, you need to approach this step-by-step.

First as outlined by @zoobab make sure the digispark is detected by the linux box. E.g. the lsusb will tell you that. 

Second make sure the i2c subsystem of the linux kernel has access to it. This may indeed require modprobing i2c-dev

Now your i2c-tiny-usb is working and everything that happens from here on is not i2c-tiny-usb specific but e.g. is the same with other i2c ports like e.g. the one on the raspberry pi.

Third make sure the pcf8574 is working on i2c. If the `i2cdetect -l | fgrep tiny` e.g gave i2c-10 as the bus id then do a

`i2cdetect -y 10`

This will give you a list of all connected i2c devices and your pcf8574 should show up. 

Finally you need to load (modprobe) the pcf8574 driver. 

Instructions like http://www.gtkdb.de/index_36_2914.html will help you once the i2c-tiny-usb is properly detected and the i2c bus can be scanned for devices.

harbaum, 2018-09-06:
This one may also be helpful https://ubuntuforums.org/showthread.php?t=1181192&p=8343003#post8343003

zoobab, 2018-09-06:
When I do an ```i2cdetect -y 8```, the dongle disseppears from the USB bus:

```
# i2cdetect -l
i2c-3	i2c       	i915 gmbus dpc                  	I2C adapter
i2c-1	i2c       	i915 gmbus vga                  	I2C adapter
i2c-8	i2c       	i2c-tiny-usb at bus 001 device 038	I2C adapter
i2c-6	i2c       	DPDDC-B                         	I2C adapter
i2c-4	i2c       	i915 gmbus dpb                  	I2C adapter
i2c-2	i2c       	i915 gmbus panel                	I2C adapter
i2c-0	i2c       	i915 gmbus ssc                  	I2C adapter
i2c-7	smbus     	SMBus I801 adapter at efa0      	SMBus adapter
i2c-5	i2c       	i915 gmbus dpd                  	I2C adapter
sabayon /dev # i2cdetect -y 8
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
sabayon /dev # lsusb
Bus 001 Device 004: ID 2232:1024 Silicon Motion 
Bus 001 Device 003: ID 8087:07da Intel Corp. 
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Maybe the firmware is crashing?

harbaum, 2018-09-06:
You are missing the pullups as stated in the readme.

harbaum, 2018-09-10:
Any progress?

zoobab, 2018-09-10:
Will retry this evening with the 2 10K resistors.

zoobab, 2018-09-11:
Will have to wait one more day, today is birthday party, so that will be for tomorrow...

closed 2018-09-15 by harbaum

zoobab, 2018-09-17:
I found this nice howto for the MCP23017:

http://elektranox.org/2017/05/0018-collabora/

PizzaProgram, 2020-03-08:
Is that patch applied from Sebastian Reichel?
Is the current driver stable?

harbaum, 2020-03-08:
The current driver is in the main kernel. You just need to check their if the patches you are interested in have been submitted and accepted there. Imho they are included.

PizzaProgram, 2020-03-08:
I could not check, that's why I've asked...

> Imho they are included.
 Than it's OK :-)