harbaum/I2C-Tiny-USB #19: Driver hangs on i2cdetect if nothing connected (digispark)

prandeamus, 2021-05-10:
I downloaded the firmware for the Digispark variant and it works fine, at least on initial testing - for example I can plug the Digispark into my Linux laptop, connect to an MCP23017 and ensure it has pullup 10K resistors, and it can detect the device (i2cdetect -y 6 in my case works fine)

However if I just plug in the digispark and let it initialise, then run i2cdetect -y <busnumber> the driver stops working altogether. Now, to be fair, I'm not expecting a meaningful result here. If you are running open collector, connecting to nothing, not even terminators, it's unreasonable to expect any data. But it seems bad to let the USB driver crash, because the only way to recover seems to be to power cycle the digispark by removing it from the USB and reinserting it.

Is this expected behaviour and is there any way to make the device at least self-heal without requiring manual power cycling? Maybe for example having a watchdog timer to force a reset if the usb polling loop breaks?

harbaum, 2021-07-09:
Have you added pullups to the i2c lines? If not the firmware will wait forever for the lines to go into high/idle state.

prandeamus, 2021-07-09:
Oh yes, if I add pull-ups it does work. I am not complaining that I2C without pullups doesn't work because they are part of the spec.

My point is that if the firmware waits "forever" the linux driver gives up (it probably has a timeout if it gets no response from the USB, which is fair). At that point, the only way out is to powercycle the ATTINY85. OK maybe you can do a RESET but if like me you are using a digispark clone the only option is to take the USB plug out and put it back again.

Would it not at least be possible to have a watchdog on the ATTINY85 and if it has absolutely no response on the bus of any kind for (say) 10 seconds, it could reset itself? That would mean it would drop into any bootloader or restart the app and there might be a chance that the system would self-heal and the linux driver would re-initialise?