harbaum/I2C-Tiny-USB #20: Adafruit Trinket M0 supported?

UweSauter, 2021-07-09:
On the [Adafruit Trinket home](https://www.adafruit.com/product/1501) is a deprecation warning, advising users to change to the [Adafruit Trinket M0](https://www.adafruit.com/product/3500). Is this also supported by this firmware?

What changes would be necessary to support the Trinket M0?

prandeamus, 2021-07-09:
I will make a guess. The original trinket uses an ATTINY85 and the M0 uses a much more sophisticated processor ATSAMD21E18. Getting to ATTINY85 to bitbang USB is an amazing feat of cleverness, but it stretches the chip to do low-speed USB and sometimes there are timing problems with individual chips or USB controllers. The ATSAMD21E18 is described as having native USB so I suppose much more is implemented in hardware & should be more reliable.

But the ATSAMD21E18  is a much different processor and therefore would require different firmware than an ATTINY85.

(If I'm wrong, I'm wrong. But that's how I read it)

harbaum, 2021-07-09:
That's correct. The i2c-tiny-usb makes extensive use of the attiny85's features.

It may be possible to write a compatible firmware for the trinket m0. But this sure needs to be written explicitly for the m0.

closed 2021-07-09 by harbaum

UweSauter, 2021-07-09:
Would you put this info into your `README.md` so that other visitors can get it while reading the project description?