harbaum/I2C-Tiny-USB #24: No precompiled binaries present

Troynica, 2023-01-25:
Hello,

The readme mentions "If in doubt use the pre-compiled firmware.hex file distributed with the source code". However, I cannot find it. Is it not there or am I overlooking something? If it is not there, would it be possible to include it?

Kind regards,
Roy

harbaum, 2023-01-26:
You are indeed right. I've added one to firmware/bin. This is untested as I don't have the flasher at hand. Please give feedback if it's working (or not).

 I've also updated the instructions to compile the firmware. This should now be easy to follow.

Troynica, 2023-01-26:
Cool, it's there. If it's built from the current source, it should be working. I will let you know if it actually does when I receive ordered parts.

Troynica, 2023-02-16:
I can confirm that it is working.
Thanks for all the effort !

Troynica, 2023-02-21:
Let me close the issue because it is solved...

closed 2023-02-21 by Troynica